mooltipass/minible #416: Unkown card

Knoxi-Code, 2023-12-06:
Hallo ich have a big problem, 
i want copie my card to a sencond full card.
Why i delete the second card profil. (first option yes and sencond option no). Than i removed the card, the mooltipass restarts and i put my first card in, to restart the clone option. But now i have two unkown cards, whats happend wrong, all data are lost ?? :(

Sazoji, 2023-12-06:
you need at least one card to clone, try restoring a database backup and see if the card was reset or the database was.
if it's reading "unknown card" you'd have to have wiped multiple cards, which would be exceedingly hard to do.
if it's showing up as that maybe it's a reading error?

Knoxi-Code, 2023-12-06:
Hmm... i have only two cards and there are unkown, and no backup. I buyed now safty reasons 10new cards :P but they need time to come.

Is there a other way the clean the card ? :/ 
But I don't think so thats a reading error, i test it again and again...

limpkin, 2023-12-06:
You may want to read our user manual to get familiar on how our device
stores your credentials (in short, the card only stores your encryption
key).
You may erase your card by bringing up the developer tab by pressing
shift-f3 in moolticute, in it you'll find the right button

Knoxi-Code, 2023-12-06:
Yes, I had a mistake in my thinking, that the credetinals on the card. 
At least I've learnt something now.

closed 2023-12-06 by limpkin