mooltipass/minible #422: Replacement with new battery does not work unplugged

ConYel, 2024-05-09:
### Expected behavior
Replace the battery and minible works unplugged.
### Actual behavior
Battery replaced but it doesn't work unplugged
### Step by step guide to reproduce the problem
?replace the battery with  a new one?
### Firmware Version
AUX MCU version:  
Main MCU version:  
Bundle version: v5 

### Operating System
- Debian Based distribution

Is there any hardware reset (security feature?) regarding the battery replacement that I haven't checked and needs to be done after the replacement?

closed 2024-05-12 by ConYel

limpkin, 2024-05-12:
hello, sorry for my delayed answer... did you manage to fix your issue?

ConYel, 2024-05-18:
Yeap, the new battery wasn't actually new so it had to be replaced by the manufacturer.